SergiuToporjinschi/OctoPrint-CalibrationTools #7: extruder not working

Therk75, 2022-02-13:
I have an ender 5 plus and after running the calibration tool once and saving changes, the extruder is not working at all. it is extruding 0 filament. I have done a restart of octoprint, a restart of the printer, I've even tried manually feeding filament with octoprint off at the printer itself and nothing is happening. Please advise.

SergiuToporjinschi, 2022-02-14:
Hi 
Did you heated the hot-end? because Marin has a prevent on cold extrusion so you will  not be able to increase steps on extruder as long as the hot-end is not... hot ... you need at least 180 If I remember correctly.

SergiuToporjinschi, 2022-02-14:
Ohh  but did you pushed the refresh button?
I will add a conditio  there

eipxxgam, 2022-02-14:
Yes, but refresh button doesn't retrieve any value.

SergiuToporjinschi, 2022-02-14:
But why is that? what Marlin version are you using. if you type in termina M92 what is the answer?

eipxxgam, 2022-02-14:
It's working M92 command in terminal. My printers firmware is Jyers 2.0.1

SergiuToporjinschi, 2022-02-14:
I have Marlin my M92 answers like this 
```
Send: M92
Recv: echo: M92 X80.0 Y80.0 Z800.0 E90.0
Recv: ok
```
Unfortunately, there is no other way to do plugins for Marlin but to send commands, listen to terminal and then parse the answer with regex.
Most probably Jyers returns the answer a bit different and in that case my plugin is not able to take the values.
If you want to help me with that I will be glad to try to support also Jyers for this will would need to know exact string that is returned on different MCommands....

eipxxgam, 2022-02-14:
Yes, there is some difference between the M92 answers

Send: M92
Recv:   M92 X80.00 Y80.00 Z400.00 E403.80
Recv: ok

SergiuToporjinschi, 2022-02-14:
yes... the answer is not starting with "echo:" 
my listeners always expects an "echo:"....

SergiuToporjinschi, 2022-02-14:
@eipxxgam I will release another version now, I have to say that I've found a common regex to detect also your type of answer but I cannot test it because I don't have Jyers  but if you want, you can update my plugin and check if refresh button is fetching the current values :) I would like to know if is working.
Thank you!

Therk75, 2022-02-15:
i typed in m92 and got this reply

Send: M92
Recv: ok

I read on marlin that i needed to use m503 if i have an older version and that yielded better results

Send: M503
Recv: echo:  G21    ; Units in mm
Recv: echo:  M149 C ; Units in Celsius
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D1.75
Recv: echo:  M200 D0
Recv: echo:Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z800.00 E100.00
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z10.00 E0.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X500 Y500 Z100 E5000
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P500.00 R1000.00 T500.00
Recv: echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 S0.00 T0.00 B20000 X8.00 Y8.00 Z0.40 E0.00
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Auto Bed Leveling:
Recv: echo:  M420 S1 Z2.00
Recv: echo:PID settings:
Recv: echo:  M301 P14.72 I0.89 D61.22
Recv: echo:Z-Probe Offset (mm):
Recv: echo:  M851 Z-2.40
Recv: ok

I think my issue may be a bit beyond the scope of your plugin, but do you have any links for showing how to upgrade my firmware, or how to upgrade marlin to the latest version? I have watched a few videos, but i dot see any mini usb ports on the side of my printer so I'm kind of grasping at straws here. or is there any other program besides octoprint i can use to change the settings on my printer.

SergiuToporjinschi, 2022-02-15:
ohh yes, I cannot see anything abnormal in your M503 report, everything seems ok. Maybe you should check the connection of stepper motor.
To upgrade your firmware for Ender 5 plus it seems you need to do it remotely. I found this video for you maybe it helps
https://www.youtube.com/watch?v=9pDoxf13_wg
In deed it seems to not be related to this plugin.
Now I will close this ticket.

closed 2022-02-15 by SergiuToporjinschi

eipxxgam, 2022-02-15:
> @eipxxgam I will release another version now, I have to say that I've found a common regex to detect also your type of answer but I cannot test it because I don't have Jyers but if you want, you can [...]

Hi @SergiuToporjinschi 
First i tried with last Jyers firmware 2.0.1 and the problem still exist. Reload button couldn't retrieve any value. 
 
Then, i reverted to the Jyers firmware 1.3.5b and i realized that the answer of M92 command consist of "echo" and it's now its working without a problem.

SergiuToporjinschi, 2022-02-15:
@eipxxgam can you update to 1.0.3? I have a confirmation that is working on Jayers on issue #13

eipxxgam, 2022-02-15:
Thank you @SergiuToporjinschi 
It's working now with Jyers 2.0.1 firmware without any issue.

SergiuToporjinschi, 2022-02-15:
I'm thinking to install Jyers 2.0.1 on my printer this weekend. I have thinking on this from Jyers 1 ... 😀