SergiuToporjinschi/OctoPrint-CalibrationTools #13: M92 code not working

ThingieBE, 2022-02-15:
Hi,

I have a Creality Ender 3 V2 printer which is connected with octoprint on a raspberry pi through USB. In the X-Y-Z steps and E-steps section when I press the buttons to get the current values the values do not appear. When I enter the M92 command in a terminal the values are returned in the terminal. The values in the steps sections stay on zero.

fnsign, 2022-02-15:
Try to refresh the page in your browser. The correct values should appear. I made the same experience when I installed and configured the plugin. I'm not sure if this is related to the plugin or to OctoPrint itself, as I make similar experiences also with other plugins (e.g. Spool Manager)

ThingieBE, 2022-02-15:
> Try to refresh the page in your browser. The correct values should appear. I made the same experience when I installed and configured the plugin. I'm not sure if this is related to the plugin or to OctoPrint itself, as I make similar experiences also with other plugins (e.g. Spool Man

ThingieBE, 2022-02-15:
Thx for the quick replies all of you. Here is the response of the M92 command:

Recv:  T:26.73 /0.00 B:26.58 /0.00 @:0 B@:0
Send: M92
Recv:   M92 X80.00 Y80.00 Z400.00 E93.00
Recv: ok
Recv:  T:26.76 /0.00 B:26.65 /0.00 @:0 B@:0

SergiuToporjinschi, 2022-02-15:
yes, but there is an update I did it yesterday can you check if you have the latest version?

ThingieBE, 2022-02-15:
> yes, but there is an update I did it yesterday can you check if you have the latest version? are you using Jyers ?

I have Calibration Tools: 1.0.2

ThingieBE, 2022-02-15:
> I just release a new version 1.0.3 I hope is working this time. Just let me know how it went. Thank you!

It is working perfect now. Lots of thanks for the quick response and the crazy fast fix and new version. You're a hero ;)