SergiuToporjinschi/OctoPrint-CalibrationTools #17: Multi extruder

mdalacu, 2022-02-16:
Hi, please make it possible to select different tools so we can calibrate our IDEX/multi extruders printers. 
Thx

SergiuToporjinschi, 2022-02-16:
Hi @mdalacu,
Unfortunately, I have just a standard Ender 3 v2 with single extruder.
But what it needs to have multi extruder? 
I'm asking because in PID autotune you can change extruder index by incrementing E value.
How you are tuning steps for extruder?
And can you provide an example of response e for `M92`? because I think that should be different in case of multi extruder.
Thank you!

mdalacu, 2022-02-16:
Hi!
Send: M92
Recv:   M92 X64.00 Y80.00 Z400.00
Recv:   M92 T0 E415.00
Recv: ok
[...]
Send: T1
Recv: echo:Active Extruder: 1
Recv: ok
[...]
Send: M92
Recv:   M92 X64.00 Y80.00 Z400.00
Recv:   M92 T1 E415.00
Recv: ok
T0 is selected as default....
thanks.

mdalacu, 2022-02-16:
So, basicly you need a new dropdown box with items T0...Tn , n beeing the number configured in Octoprint, printer profiles as "Number of Extruders". And on change...send Tn command before reading M92 or setting something.
Thanks

SergiuToporjinschi, 2022-02-16:
oohh so that's completely different ...
I cannot promise anything yet... I working on this on my spare time

SergiuToporjinschi, 2022-02-16:
> So, basicly you need a new dropdown box with items T0...Tn , n beeing the number configured in Octoprint, printer profiles as "Number of Extruders". And on change...send Tn command before reading M92 or setting something. Thanks

yes but is not so simple because I need the initial E value from M92 and there is another line involved in the printer response ...

mdalacu, 2022-02-16:
Which one? The response contains only 2 linesRecv:   M92 X64.00 Y80.00 Z400.00
Recv:   M92 T0 E415.00