SergiuToporjinschi/OctoPrint-CalibrationTools #23: M303 (PID Tuning) 'U1' option doesn't work for Prusa Printers

Naugrimohtar, 2022-02-18:
Hello,

I have a MK2.3 and a MK3 Prusa printer. The 'U1' argument for the M303 does not appear to be implemented for these printers. as a result the new PID values determined by this plugin are not saved/updated. I recommend either adding a comment so other used don't expect that it works for them - or implementing different functionality that allows for reading of the discovered paramters, setting, and saving of them.

reference: https://help.prusa3d.com/en/article/prusa-specific-g-codes_112173
Look for M303, note there is no mention of the 'U1' option like there is for Marlin.
Via my own testing I can confirm that it is not saving the values after execution.

SergiuToporjinschi, 2022-02-19:
Hi @Naugrimohtar,
Pfff I don't have a Prusa, I have and Ender 3V2, so I I don't know if I can help you but can you tel me how you are running PID autotune for Prusa? I would need the command to start the process and the command to take current value also I would need the exact answer from your printer. Maybe I can make it work for Prusa also.

Naugrimohtar, 2022-02-20:
It’s the same command to start the tuning (without the U1)

Once the values are found, I note what the final values were reported as and then use these commands to save them.

----Extruder----
M303 E0 S215 C8;run
M301 P# I# D# ;set
M301 P17.36 I1.33 D56.52
M500

----Heatbed----
M303 E-1 S60 C8
M304 P# I# D#
M304 P82.14 I4.23 D398.68
M500

SergiuToporjinschi, 2022-02-20:
Aaah ok, I will see what I can do. 
U1, in Marlin, is saving the values directly in memory and is easy for me to just do an M500
Should not be to hard to change. Unfortunatly i'm doing this in my spare time.
Thanks for info.