SergiuToporjinschi/OctoPrint-CalibrationTools #25: Error on Starting Extrusion

MisterTFM, 2022-02-28:
**Describe the bug**
After pre-heating nozzle to 200C and then clicking the "Start extrusion" button, a pop-up occurs that states: "error on starting extrusion, an error cured"

Why does the error message appear?

**Versions:**
 - OctoPrint version[1.7.3]: 
 - Browser [chrome]:
 - Printer type: Ender 3 Pro
 - Printer firmware [displayed message in `M115` command]:

Marlin bug fix 2.0.X

SergiuToporjinschi, 2022-02-28:
Hi,
hmm something happened also on front-end can you send me the JS logs also ? or network requests?

SergiuToporjinschi, 2022-02-28:
aaa I think you are using a different python version can you tell me what Python version are you using?

MisterTFM, 2022-02-28:
Python 3.10.2

Sent from iphone.

MisterTFM, 2022-02-28:
Enabled log for Calibration Tools
[octoprint-2-28-22.log](https://github.com/SergiuToporjinschi/OctoPrint-CalibrationTools/files/8154815/octoprint-2-28-22.log)