SergiuToporjinschi/OctoPrint-CalibrationTools #26: Heated bed tuning not working

Sebe47, 2022-03-03:
No message in Terminal for starting PID bed Tuning

Send: M303 C20 E-1 S60 U1
Recv: PID Autotune start

Only if i send the command manually 

Hotend PID Tuning is working for my Printer

tiagomaricate, 2022-03-04:
Here nothing is working.

SergiuToporjinschi, 2022-03-05:
Hi @Sebe47, 
you know that you may have octoprint set for not populating terminal if is not open you need to open two tabs in that case, To have one with terminal and one with this plugin.
Can you help me with python version and marlin version ? and some logs would be nice.

skl111, 2022-04-08:
what logs do you need exactly?

skl111, 2022-04-08:
Firmware Name: Marlin bugfix-2.0.x (Feb 6 2022 20:52:55)
Source Code Url: github.com/MarlinFirmware/Marlin
Protocol Version: 1.0
Machine Type: Ender 5 Pro BTT SKR Mini E3 V2.0
the HE PID Tune works but the Bed does not. Pressing the button does not even bring up the pop up or give and error message