SergiuToporjinschi/OctoPrint-CalibrationTools #27: After upgrading to Marlin 2.0.9.3 - Values not pulled from EEPROM

brandonclyon, 2022-03-05:
I flashed from Marlin 2.0.7.2 to 2.0.9.2 yesterday.

Since then the plugin will not automatically pull from EEPROM, checking terminal the commands are issues and results given but the are not imported into the plugin interface.

SergiuToporjinschi, 2022-03-05:
Hi @brandonclyon, What do you mean by that, when you push refresh buttons?

brandonclyon, 2022-03-05:
Correct, in the webUI I click to pull the current values for either PID or eStep, the terminal shows that the command is issued and gives a response but those values are not transferred to the plugin fields.

Loukey42, 2022-03-13:
Similarly, I had it working perfectly with Creality motherboard 4.2.2 running stock firmware then Creality 4.2.7 running both stock and custom (2.0.9.3 and 2.0.bugfix) but just changed over to a SRK E3 mini v2 running 2.0.9.3 and it will not pull any values.  I've tried a few times and can't get the plugin to pull any values.  It runs the update correctly.  I can supply any logs which could be helpful.

skl111, 2022-04-08:
Firmware Name: Marlin bugfix-2.0.x (Feb 6 2022 20:52:55)
Source Code Url: github.com/MarlinFirmware/Marlin
Protocol Version: 1.0
Machine Type: Ender 5 Pro BTT SKR Mini E3 V2.0
the HE PID Tune works but the Bed does not. Pressing the button does not even bring up the pop up or give and error message

jthomeo, 2022-12-21:
Any updates on this issue? How the plugin get the data to populate it's fields?

IsmaelPR1977, 2023-01-20:
Same issue, Rambo Mini 1.3a Prusa MK2.5, values will not load nor do I have the option to refresh/pull values from board.

netmilk, 2023-12-07:
Hotfixed in my PR #36. [Here](https://github.com/netmilk/OctoPrint-CalibrationTools/archive/refs/heads/main.zip) you can download the .`zip` of my fork and install it manually until it's merged.