SergiuToporjinschi/OctoPrint-CalibrationTools #31: Python Error: AttributeError: module 'collections' has no attribute 'Callable'

hseibel12, 2022-08-27:
After installing the CalibrationTools on my OPiZ I received an error message box: 
"Error on starting extrusion
An error curred"

When looking to octoprint.log I found following message:

2022-08-27 21:17:17,002 - octoprint.plugins.CalibrationTools - ERROR - Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/lib/python3.10/site-packages/octoprint_CalibrationTools/api.py", line 46, in on_api_command
    return api["cls"].apiGateWay(self, command, data)
  File "/home/pi/OctoPrint/venv/lib/python3.10/site-packages/octoprint_CalibrationTools/EStepsApi.py", line 33, in apiGateWay
    self.registerGCodeWaiter("M92", self.m92GCodeResponse, m92Event)
  File "/home/pi/OctoPrint/venv/lib/python3.10/site-packages/octoprint_CalibrationTools/hooks.py", line 91, in registerGCodeWaiter
    self.registerRegexMsg(reg, func, *arguments)
  File "/home/pi/OctoPrint/venv/lib/python3.10/site-packages/octoprint_CalibrationTools/hooks.py", line 94, in registerRegexMsg
    if regex is None or func is None or not isinstance(func, collections.Callable):
AttributeError: module 'collections' has no attribute 'Callable'

**Versions:**
 - OctoPrint version 1.8.2: 
 - Python V3.10
 - Browser Firefox:
 - Printer type: Ender3
 - Printer firmware [displayed message in `M115` command]:

Send: M115
Recv: FIRMWARE_NAME:Marlin 2.0.6 (Feb 10 2022 17:22:16) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender-3 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:BINARY_FILE_TRANSFER:0
Recv: Cap:EEPROM:1
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:1
Recv: Cap:RUNOUT:1
Recv: Cap:Z_PROBE:1
Recv: Cap:LEVELING_DATA:1
Recv: Cap:BUILD_PERCENT:0
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:1
Recv: Cap:PROMPT_SUPPORT:0
Recv: Cap:SDCARD:1
Recv: Cap:AUTOREPORT_SD_STATUS:0
Recv: Cap:LONG_FILENAME:0
Recv: Cap:THERMAL_PROTECTION:1
Recv: Cap:MOTION_MODES:0
Recv: Cap:ARCS:1
Recv: Cap:BABYSTEPPING:1
Recv: Cap:CHAMBER_TEMPERATURE:0
Recv: ok
Send: M155 S2
Recv: ok

jcoetzee1188, 2023-06-25:
Is there a fix for this? I have the same problem.

Thanks

corpeder, 2023-06-26:
Im pretty sure this is due to the collections module being updated in python 3.10. I wonder if a python downgrade would work?